soonuse/epd-library-arduino #2: Bug in epdif.cpp

psitech, 2018-04-02:
In epd-library-arduino/2.9inch_e-paper/arduino/libraries/epdif.cpp, the following 2 lines must be swapped:
60   SPI.beginTransaction(SPISettings(2000000, MSBFIRST, SPI_MODE0));
61   SPI.begin();
otherwise the SPI bus won't work [as I found out when trying to get the 2.9" display to work].
So the correct sequence is:
60   SPI.begin();
61   SPI.beginTransaction(SPISettings(2000000, MSBFIRST, SPI_MODE0));

Regards,
Paul

closed 2018-08-30 by soonuse

psitech, 2019-03-24:
Hi Yehui,
Thanks for fixing the libraries.
However, the downloadable sources on the Waveshare website have not yet been updated with this fix.
Do you also take care of that?

Thanks,
Paul

soonuse, 2019-03-28:
All right, it's updated now.