soonuse/epd-library-arduino #11: library.properties file missing

badrihippo, 2019-04-09:
Starting version 1.5.x, Arduino requires a new `library.properties` file which is used by the Library Manager to import those libraries. Details can be found at [this wiki page](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification).

The various e-Paper libraries in this project are missing `libraries.properties`; it would help to have them in order to easily upload and manage them with Arduino's new tools. Details can be found on the aforementioned page, but here is a sample file we could use:

```
name=epd4in2
version=1.0
author=Yehui from WaveShare
maintainer=Yehui from WaveShare
sentence=Arduino driver for the WaveShare 4.2inch e-paper display
paragraph=You can find more information on this and other displays at https://www.waveshare.com/wiki/Main_Page#OLEDs_.2F_LCDs
category=Display
url=https://github.com/soonuse/epd-library-arduino
```

If you give me a final sample file you want to use, I can generate `libraries.properties`  for all the libraries and submit them in a pull-request. (Unfortunately, you will have to change the version every time you make an update, which means updating all those various files once again).

soonuse, 2019-04-25:
I probably won't provide a library.properties file since the authors and maintainers of the new e-paper series are not only me. Although I prefer to write 'Waveshare' as the author and maintainer, but my coworker has created a dedicated repository under Waveshare https://github.com/waveshare/e-Paper. Thank you for your support.

badrihippo, 2019-04-25:
OK...should I raise the issue in the new repository? That doesn't have `library.properties` either, so will not work in newer Arduino interfaces.

soonuse, 2019-04-26:
OK. For ease to use, library.properties

closed 2019-04-26 by soonuse