soonuse/epd-library-arduino #16: Add newer displays

mousaka20, 2021-04-21:
By now Waveshare has many more displays available. And their [wiki](https://www.waveshare.com/wiki/5.65inch_e-Paper_Module_(F)) provides more information and resources.

I personally would like the 5.65in to be added.

Any chance more recent display are added?

madewhatnow, 2021-06-13:
I agree - several of the new larger display (7.5" V2) are missing.

idimou, 2022-11-02:
Also WEMOS updated their LOLIN epaper 2.13" to use SSD1680 controller. Any chance we can add that?

weckerle, 2025-08-28:
Support for waveshare's 13,3 7-color epaper display would also be very interesting. Or does anyone know a way how to get it working atm?
https://www.waveshare.com/product/displays/e-paper/13.3inch-e-paper-hat-plus-e.htm